norberts1/hometop_HT3 #2: Ems buderus?

delchrys, 2017-06-30:


delchrys, 2017-06-30:
Hmm forgot the rest sorry.
Does this work or can be adapted to work with ems buderus also?
Like to know which parameters I can get out of my nefit heater

norberts1, 2017-07-01:
I'll think you can use the Hardware (HT3_Mini- and HT3_Micro-Adapter) without modifications also for the EMS-Bus used by Buderus. The pysical layer is the same between Junkers- and Buderus-Bus.
Also most of the things on the higher layers are compare-able but not identical.
So it is required to modify the decoder for the received telegrams and modifications are also necessary for transmitting data to the heater-bus.
There are projects already working on the **Buderus** EMS-Bus, see URL's:
https://emswiki.thefischer.net/doku.php?id=wiki:ems:net_io#webseite
https://github.com/maniac103/ems-collector/tree/master/collector

delchrys, 2017-07-01:
Ok thanks I knew already of the buderus forum threads. Actually I'm looking to connect something to the gateway which will upload my data to my domoticz server. So a arduino or esp8266 directly at the bus gateway. It will decode the messages and update some values in my domoticz application.
I prefer a wifi connection because of the distance between my heater and my server. 
So I want 
Heater --> ems bus gateway --> arduino or esp8266 --> domoticz server RPi
Where the arduino does the decoding of the bus messages. And makes a get request to my server which updates some values.

norberts1, 2017-07-02:
For this purpose you can use another adapter based on the ESP8266. 
You can make a direct link between your heater and domoticz Server like:
 Heater --> esp8266 -->  domoticz server RPi
The data-transfer is as streaming-data without decoding. This must be done on your server.
But the source is also available and included as java-source in that application.
See URL's:
  https://github.com/susisstrolch/EMS-ESP12
and
  https://www.mikrocontroller.net/topic/356289#new (german language)

delchrys, 2017-07-02:
Yes was also tinkering with that. But since it does not sent anything it's hard to adapt to my needs. It only shows some info in the debugging console but not sending anything. That part has to be rewritten completely.

norberts1, 2017-08-20:
Hi,
I have made a test for that board on my heater-system.
With one minor hardware-change it works so far also on the junkers-heaterbus.
Also the received data is available with that call:
  netcat -w 30 192.168.254.123 23 >>ems-$(date +%s).log
The data-rx at port 23 is logged to one file.
But anyway the data-content has special addons for that handing with the buderus-collector software and is not direct useable for the junkers telegramm-handling.
Result is:
The hardware can be used on the junkers heater-systembus (Heatronic/EMS2) but the software has to be rewritten.
(board-description see: https://www.mikrocontroller.net/topic/356289#new)

closed 2018-11-03 by norberts1