norberts1/hometop_HT3 #4: ht_proxy exceptions lead to  RuntimeError: can't start new thread

timmmmmey, 2018-04-04:
Hi,

recently some exceptions occur in ht_rpxy 

> `Apr 04 19:58:07 fhempi ht_proxy.py[2112]: Exception happened during processing of request from ('127.0.0.1', 43732)
Apr 04 19:58:07 fhempi ht_proxy.py[2112]: ----------------------------------------
Apr 04 19:58:07 fhempi ht_proxy.py[2112]: ----------------------------------------
Apr 04 19:58:07 fhempi ht_proxy.py[2112]: Exception happened during processing of request from ('127.0.0.1', 43734)
Apr 04 19:58:07 fhempi ht_proxy.py[2112]: ----------------------------------------
Apr 04 19:58:07 fhempi ht_proxy.py[2112]: ----------------------------------------
Apr 04 19:58:07 fhempi ht_proxy.py[2112]: Exception happened during processing of request from ('127.0.0.1', 54694)
Apr 04 19:58:07 fhempi ht_proxy.py[2112]: ----------------------------------------
Apr 04 19:58:07 fhempi ht_proxy.py[2112]: ----------------------------------------
Apr 04 19:58:07 fhempi ht_proxy.py[2112]: Exception happened during processing of request from ('127.0.0.1', 54696)
Apr 04 19:58:07 fhempi ht_proxy.py[2112]: ----------------------------------------
Apr 04 19:58:07 fhempi ht_proxy.py[2112]: ----------------------------------------

Apr 04 19:58:07 fhempi ht_proxy.py[2112]: Exception happened during processing of request from ('127.0.0.1', 55762)
Apr 04 19:58:07 fhempi ht_proxy.py[2112]: ----------------------------------------
Apr 04 19:58:07 fhempi ht_proxy.py[2112]: ----------------------------------------
Apr 04 19:58:07 fhempi ht_proxy.py[2112]: Exception happened during processing of request from ('127.0.0.1', 37414)
Apr 04 19:58:07 fhempi ht_proxy.py[2112]: ----------------------------------------
Apr 04 19:58:07 fhempi ht_proxy.py[2112]: ----------------------------------------
Apr 04 19:58:07 fhempi ht_proxy.py[2112]: Exception happened during processing of request from ('127.0.0.1', 37416)
Apr 04 19:58:07 fhempi ht_proxy.py[2112]: ----------------------------------------
`

which leads to the following error:
`Apr 04 20:18:07 fhempi ht_proxy.py[2112]: ----------------------------------------  File "lib/ht_proxy_if.py", line 409, in add_client
Apr 04 20:18:07 fhempi ht_proxy.py[2112]: txThread.start()
Apr 04 20:18:07 fhempi ht_proxy.py[2112]: File "/usr/lib/python3.4/threading.py", line 850, in start
Apr 04 20:18:07 fhempi ht_proxy.py[2112]: _start_new_thread(self._bootstrap, ())
Apr 04 20:18:07 fhempi ht_proxy.py[2112]: RuntimeError: can't start new thread
Apr 04 20:18:07 fhempi ht_proxy.py[2112]: Traceback (most recent call last):
Apr 04 20:18:07 fhempi ht_proxy.py[2112]: File "/usr/lib/python3.4/socketserver.py", line 613, in process_request_thread
Apr 04 20:18:07 fhempi ht_proxy.py[2112]: self.finish_request(request, client_address)
Apr 04 20:18:07 fhempi ht_proxy.py[2112]: File "/usr/lib/python3.4/socketserver.py", line 344, in finish_request
Apr 04 20:18:07 fhempi ht_proxy.py[2112]: self.RequestHandlerClass(request, client_address, self)
Apr 04 20:18:07 fhempi ht_proxy.py[2112]: File "/usr/lib/python3.4/socketserver.py", line 669, in __init__
Apr 04 20:18:07 fhempi ht_proxy.py[2112]: self.handle()
Apr 04 20:18:07 fhempi ht_proxy.py[2112]: File "lib/ht_proxy_if.py", line 445, in handle
Apr 04 20:18:07 fhempi ht_proxy.py[2112]: _ClientHandler.add_client(self._myownID, self.request)
Apr 04 20:18:07 fhempi ht_proxy.py[2112]: File "lib/ht_proxy_if.py", line 409, in add_client
Apr 04 20:18:07 fhempi ht_proxy.py[2112]: txThread.start()
Apr 04 20:18:07 fhempi ht_proxy.py[2112]: File "/usr/lib/python3.4/threading.py", line 850, in start
Apr 04 20:18:07 fhempi ht_proxy.py[2112]: _start_new_thread(self._bootstrap, ())
`
maybe we should clone the thread on an exception or use a threadpool? Is there any possibility to get more infomation about the exception?

norberts1, 2018-04-04:
Hi timmmmey,
it looks like a memory-leak. Reason could be the many connection - requests of socket-clients.
Please look for that portnumber: 8088 in your system. It could be that it is already used for something else.
So change that port-numer (for server and client) in the configuration-File: **~/HT3/sw/etc/config/ht_proxy_cfg.xml** to any other unused port-number and start the Pi again.

timmmmmey, 2018-04-05:
Hi Norbert,

the port is only used by ht_proxy and and the only connection initiated by the fhem heatronic plugin. Changing the port didn't help (as long I change it in fhem as well). 

Every ~10 minutes there is also a short time connection from ht_netclient (cronjob, because fhem plugin sending temperatures is not working well and commands from the ht_netclient sometimes got ignored from the heater, so I put this in a cron job)

Your memory-leak theorie seems to fit, my ht_proxy is occupying 1,6GB virtual memory right now. I know virtual memory is not the real used memory, so I will try to run it with valgrid/massif and post the results afterwards.

timmmmmey, 2018-04-30:
`--------------------------------------------------------------------------------
Command:            python3 /opt/ht3/ht_proxy.py
Massif arguments:   --massif-out-file=/opt/ht3/mem_prof
ms_print arguments: /opt/ht3/mem_prof
--------------------------------------------------------------------------------

    MB
1.158^                     #
     |                     #
     |                     #:::::::::::::@@@::::::::::::::::::@::::@:::::@::::
     |                 :  @#::: ::: : :  @  : :: ::: :::::::::@: ::@:::::@::::
     |                 :::@#::: ::: : :  @  : :: ::: :::::::::@: ::@:::::@::::
     |              ::::::@#::: ::: : :  @  : :: ::: :::::::::@: ::@:::::@::::
     |              :: :::@#::: ::: : :  @  : :: ::: :::::::::@: ::@:::::@::::
     |              :: :::@#::: ::: : :  @  : :: ::: :::::::::@: ::@:::::@::::
     |             ::: :::@#::: ::: : :  @  : :: ::: :::::::::@: ::@:::::@::::
     |            :::: :::@#::: ::: : :  @  : :: ::: :::::::::@: ::@:::::@::::
     |       : @:::::: :::@#::: ::: : :  @  : :: ::: :::::::::@: ::@:::::@::::
     |       ::@:::::: :::@#::: ::: : :  @  : :: ::: :::::::::@: ::@:::::@::::
     |       ::@:::::: :::@#::: ::: : :  @  : :: ::: :::::::::@: ::@:::::@::::
     |      :::@:::::: :::@#::: ::: : :  @  : :: ::: :::::::::@: ::@:::::@::::
     |      :::@:::::: :::@#::: ::: : :  @  : :: ::: :::::::::@: ::@:::::@::::
     |    :::::@:::::: :::@#::: ::: : :  @  : :: ::: :::::::::@: ::@:::::@::::
     |  @::::::@:::::: :::@#::: ::: : :  @  : :: ::: :::::::::@: ::@:::::@::::
     |  @::::::@:::::: :::@#::: ::: : :  @  : :: ::: :::::::::@: ::@:::::@::::
     |  @::::::@:::::: :::@#::: ::: : :  @  : :: ::: :::::::::@: ::@:::::@::::
     | :@::::::@:::::: :::@#::: ::: : :  @  : :: ::: :::::::::@: ::@:::::@::::
   0 +----------------------------------------------------------------------->Mi
     0                                                                   368.9

Number of snapshots: 67
 Detailed snapshots: [2, 9, 19, 20 (peak), 29, 46, 50, 60]

--------------------------------------------------------------------------------
  n        time(i)         total(B)   useful-heap(B) extra-heap(B)    stacks(B)
--------------------------------------------------------------------------------
  0              0                0                0             0            0
  1      7,910,915          118,320          115,806         2,514            0
  2     12,608,269          283,384          279,722         3,662            0
98.71% (279,722B) (heap allocation functions) malloc/new/new[], --alloc-fns, etc.
->53.11% (150,496B) 0x690CA: ??? (in /usr/bin/python3.4)
|
->13.65% (38,686B) 0x83142: PyBytes_FromStringAndSize (in /usr/bin/python3.4)
|
->12.52% (35,472B) 0xDBCFA: _PyBytes_Resize (in /usr/bin/python3.4)
|
->05.74% (16,270B) 0x6014A: PyUnicode_New (in /usr/bin/python3.4)
|
->05.10% (14,448B) 0x5E856: PyDict_New (in /usr/bin/python3.4)
|
->04.26% (12,062B) in 47 places, all below massif's threshold (01.00%)
|
->02.89% (8,192B) 0xC76E2: _PyFaulthandler_Init (in /usr/bin/python3.4)
|
->01.45% (4,096B) 0xD350A: ??? (in /usr/bin/python3.4)

--------------------------------------------------------------------------------
  n        time(i)         total(B)   useful-heap(B) extra-heap(B)    stacks(B)
--------------------------------------------------------------------------------
  3     17,035,122          262,976          258,584         4,392            0
  4     24,699,125          339,792          333,949         5,843            0
  5     29,400,944          361,952          355,150         6,802            0
  6     34,929,209          469,112          461,906         7,206            0
  7     41,417,527          616,424          607,192         9,232            0
  8     46,100,465          604,240          594,325         9,915            0
  9     51,661,824          642,496          631,829        10,667            0
98.34% (631,829B) (heap allocation functions) malloc/new/new[], --alloc-fns, etc.
->54.07% (347,392B) 0x690CA: ??? (in /usr/bin/python3.4)
|
->11.42% (73,392B) 0x6014A: PyUnicode_New (in /usr/bin/python3.4)
[... 229 lines cut ...]
|
->03.09% (34,496B) 0x5E856: PyDict_New (in /usr/bin/python3.4)
|
->02.57% (28,672B) 0xD350A: ??? (in /usr/bin/python3.4)
|
->01.50% (16,704B) 0xB328A: ??? (in /usr/bin/python3.4)
|
->01.47% (16,384B) 0x103ECE: ??? (in /usr/bin/python3.4)
|
->01.24% (13,825B) 0xAE812: ??? (in /usr/bin/python3.4)
|
->01.05% (11,712B) 0x9AFAE: _PyDict_NewKeysForClass (in /usr/bin/python3.4)

--------------------------------------------------------------------------------
  n        time(i)         total(B)   useful-heap(B) extra-heap(B)    stacks(B)
--------------------------------------------------------------------------------
 61    368,491,931        1,115,928        1,095,480        20,448            0
 62    372,163,591        1,115,904        1,095,464        20,440            0
 63    375,849,820        1,115,928        1,095,468        20,460            0
 64    379,525,175        1,115,904        1,095,464        20,440            0
 65    383,196,627        1,115,936        1,095,480        20,456            0
 66    386,871,544        1,116,632        1,096,168        20,464            0

`
Scheint ein memory-leak zu sein. Jedoch nutzt definitiv nichts anderes als das FHEM-Plugin und der ht_netclient (wobei der häufig gestartet wird ~ alle 10 Minuten). Irgendwie scheint die Verbindung nicht richtig zu terminieren und daher die Ressourcen nicht freigegeben oder so. Kann ich noch irgendwas prüfen?

norberts1, 2018-05-01:
Hi timmmmmey,
it seems to be a bit strange for me. This lack of memory behaviour running the 'ht_proxy.server' I haven't seen on my maschine. On my RPI B it runs years so far. On other maschines too.
To fix that problem on your maschine please do following:
1.Deactivate the HT3-applikation under your folder /opt. It runs there as user 'root' and that environment could be a bit different.
2. Deaktivate the fhem-applikation or stop it  by hand.
3. Install the HT3-Applikation again as **user: pi** (git clone....)
4. Move the folder ~/hometop_HT3/HT3/ to ~/HT3.
5. Install the startscripts using sudo as defined in documentation (only this as user: root)
6. Modify the config-file(s ) 'ht_proxy_cfg.xml'  for your used serial ht_adapter (baudrate, portname etc must be modified. This depends on your system and the adapter.
7. create the databases (only if you are using them).
8. restart the RPi again and stop that running fhem.
Let's run a time wating for any wrong behaviour.
After all you can check the receiving of Heaterbus-data with:
 http://<IP-adr of your RPI>:8088
The data are depending of the ht_adapter and heater-system, see picture.
[image]

Let me ask: 
Which type of RPI you are using?
Which type of ht_adapter you are using?

norberts1, 2018-05-01:
empty

closed 2018-11-03 by norberts1