norberts1/hometop_HT3 #5: Simple question

Mhaluch, 2018-05-22:
Hi,

I would like to integrate the Junkers just for reading the values from the heather. I do not want to modify anything remotely.

Do I need to build the solution with ht-transceiver or just combination of RPI 3 with HT3-Miniadapter is enough?

DO I have a correct understanding that ht-transceiver is needed for controlling the heater but HT3-Miniadapter is only for reading??

Thanks for answer.

norberts1, 2018-05-22:
Hi,
yes you are wright. The ht_transceivers (ht_pitiny or ht_piduino) are only required for controlling the heater.
For your purposes (only receiving of bus-data) the HT3_mini-Adapter or the HT3_micro-Adapter are OK.

Mhaluch, 2018-05-24:
Hi,

Thank you for Your reply. I am waiting now for good delivery and I will start to build the connection with Junkers. My goal is to connect RPI with Vera system (like Domoticz, Fibaro) and to provide to Vera the status of heather. 

Can I count on Your further help in case of difficulties?

What is Your recommendation how to do this integration??
- HTTP query from Vera to RPI
- SQL query from Vera to RPI
- other solutions?

THanks!

norberts1, 2018-05-25:
Hi,
* HTTP query from Vera to RPI -->> isn't supported.
* SQLite database -->> possible, if enabled.
* MQTT dataexchange -->> possible if enabled and mqtt-broker is running.
* SPS-commands -->> SPS-IF to get decoded data from ht_collgate.py (if enabled).
For above purposes you must run at least ht_proxy.server, ht_collgate.py and any of above IFs to get that decoded heater-data.

closed 2018-11-03 by norberts1