norberts1/hometop_HT3 #6: Serial0 access device by Coollgate.run

Mhaluch, 2018-05-29:
Hi,

I have build a device, connect it to heater and RPI but I have an issue with /dev/serial0 access.

I have done all the changes in Stretch configuration files which are mentioned in the threat for RPI 3b to make the Serial0 available only for serial connection. 

But still I am getting below error.

29.05.2018 23:15:12 INFO: Starting 'Ccollgate.run()
29.05.2018 23:15:12 CRITICAL: cht_if_worker();Error;couldn't open requested device:/dev/serial0
29.05.2018 23:15:12 CRITICAL: ccollgate().run();Error;could not start 'ht-interface' with file:'./etc/config/HT3_db_cfg.xml'
29.05.2018 23:15:12 CRITICAL: ccollgate().run();Error; terminated

@norberts1 , can you help and suggest what else should be changed in configuration?

norberts1, 2018-06-01:
Hi,
on my RPI3b the /dev/serial0 and /dev/serial1 are defined as links.
/dev/serial0 -> ttyS0
/dev/serial1 -> ttyAMA0
Please check this in your /dev - folder and use /dev/serial1 instead.
Or you can use /dev/ttyAMA0 direct. Be shure to have that 'dialout' set for this device and user.

Mhaluch, 2018-06-01:
Hi,
Thanks for answer, but still it does not work. 
It is strange but I was able to make it working in SOCKET mode, but not in ASYNC mode. 
In SOCKET mode everything works perfectly, but in ASYNC mode I still have this error.
Additionally MQTT integration with Vera was succesfull and now all parameters are published also to Vera. 
Great software :)
BTW,
Do you know maybe if it is possible to read the gas (fuel) consumption by Junkers per day or per hour?
Is it provided by Junkers via CAN-BUS to RPI??
If yes, which parameter stands for it?

norberts1, 2018-06-02:
Hi,
per default **only** the ht_proxy-server is connected to the hardware (ht_adapter) using that serial device. Any other running modul must be connected as client to that ht_proxy-server (Socket.connection).
You can reconfigure this and connect that ht_collgate to that serial port and ht_adapter. But then you have to disable the ht_proxy-server doing the same thing.
Anyway this seems to be not very usefull, cause then you can't connect any other ht_client to the ht_proxy-server for receiving raw heater-data.
I'm searching for telegrams on the heater-bus for any additional informations like 'gas-consumption' etc. But this is strongly dependant from that heater-device you are using.
In my heater-system I didn't find yet this information and the attached telegrams. But I'm still searching for.
The Heatronic or EMS2 isn't a CAN-Bus, it is a 2-wire bus bearing the power and the signalling for the connected heater-moduls.

Mhaluch, 2018-06-02:
My heater is CerapurSmart ZSB 22-3 with Heatronic 3. Do you have similar?
Where can I find such telegrams which you have attached together with software?

norberts1, 2018-06-06:
My heater is CSW24 with Heatronic3 and regarding yours ZSB 22-3 I can't say much more then is already written in the manual. Please look in you manuals for finding sensors in you heater. But I can say that in my heater there isn't build in a flow-sensor for gas-consumption.
Once again I have searched for telegrams and didn't find any telegrams for 'gas-consumption' in the heatronic3 telegrams. Perhaps the newer EMS2 bus and some of the new heaters have any sensors and telegrams available.
Anyway there are telegrams available with flow-control of water and also for that preasure. But this sensors aren't build in in my heater.
The telegram-description (in german language) you can find under folder: 
  ~/HT3/sw/etc/html/HT3-Bus_Telegramme.html
Telegram with MesID:52 is bearing that water-flow message. But not all telegrams are included in this documentation.
If you like you can check your system with that HT3_Analyser regarding that telegrams. Perhaps you can find some of the defined telegrams.
If there are some unknown once let me know ore you an create a binary logfile bearing the telegrams from your heater-system for further investiation.

Mhaluch, 2018-06-09:
Hi,

Thanks for the info. I have tried to run the HT3_Analyser but I go the following error.

Traceback (most recent call last):
  File "./HT3_Analyser.py", line 35, in <module>
    HT3_Analyser.run()
  File "lib/ht3_worker.py", line 205, in run
    GUI = gui_worker.gui_cworker(ht3_cworker._gdata, self.__hexdump_window, self.__gui_titel_input, logger=self._logging)
  File "lib/gui_worker.py", line 104, in __init__
    self.__main = tkinter.Tk()
  File "/usr/lib/python3.5/tkinter/__init__.py", line 1880, in __init__
    self.tk = _tkinter.create(screenName, baseName, className, interactive, wantobjects, useTk, sync, use)
_tkinter.TclError: no display name and no $DISPLAY environment variable

Should it be run through SSH or GUI enviroment?

Mhaluch, 2018-06-09:
OK, In GUI works :)

Mhaluch, 2018-06-09:
VNC over Ethernet :)

closed 2018-11-03 by norberts1