norberts1/hometop_HT3 #8: Would this work with Bosch heater ?

donnib, 2018-10-30:
Hi,
I have a Bosch gas heater with weather compensation FW100 (i think it uses Heatronic 3). Can this be used with that ? If so what kind of hardware would i need ? I could probably use an ESP8266 to connect to the heather if it's possible.

/donnib

norberts1, 2018-11-01:
Hi donnib, if your heater has the brand 'Bosch' and it works with FW100 then it has the right protokoll to work with Heatronic and this project.
You can use any of that adapters designed for this project if you want to receive telegrams. If you want to send /command the heater, then you need the ht_pitiny- or ht_piduion-adapter.
You should never connect the ESP8266 direct to the heater-bus, cause it needs to mutch current.
I have checkt one board additional with Stepdown-converters and it takes at least 40mA current average from the heater-bus.
This value is critical and should not be used for direct current consumtion from the heater-bus.
You can use the ESP8266 with separate power and opto-isolation, but anyway the software has to be written for that design.
If you are able to build that adapters from this project then you should use them.

norberts1, 2018-11-01:
The 'mini-adapter' is the best one for your purposes to receive ht_busdata.
It's designed for the RPi and uses only same diodes and one optocoppler.
For the 'micro-adapter' you need an additonal USB-UART converter and a motherboard for the adapter.
Yes, MQTT is configurable. Default it is disabled, but it can be enabled in the 'collgate_cfg.xml' configurationfile.
The MQTT broker-address is also configurable in the 'mqtt_client_cfg.xml' configurationfile.
It's default address: localhost has to be changed then to the current broker-server.
See config-file: 'mqtt_client_cfg.xml'
  
    <mqtt_broker>
        <serveraddress>**localhost**</serveraddress>
        <portnumber>1883</portnumber>   
        <user></user>
        <password></password>
    </mqtt_broker>

closed 2019-11-22 by norberts1