norberts1/hometop_HT3 #9: Problem with reporting CO, CW works OK

Mhaluch, 2019-01-04:
Hi,

Since yesterday I can experience very strange results using RPI + HT3 together with Junkers heater.

Temp of water is reported correctly only in case the tap water is beying heat. Additionally all the others information like pomps are also OK.

In case the water is heat for central heating is not reported at all.

Can you please guide me where to look for the problem. Is it hardware (mean RPI and circuit boards) or Junkers itself.

Very strange. 

I am attaching the picture. I did the translation to polish language of the grapg but th colous are not changed.
[image]

Mhaluch, 2019-01-05:
Hi,

Thank you for feedback.

At this time there is no any ECO mode set up. Heating works as regular.
It just happened today again. When I checked the Ccolegate.log I found that some information are missing exactly within the "problematic" period.

I mean below. I am wondering that this is a root cause of this issue. I am not sure what below info means but I guess it is important :)

The last info before the problem on the graph exist:
05.01.2019 08:58:28 DEBUG:   24_0 ;Byte10:19;T2-buffer:3276.8;I-current:6553.5;pressure:255;WW-flow:0;Byte27:0;Byte28:0

The first info after:
05.01.2019 11:11:30 DEBUG:   24_0 ;Byte10:19;T2-buffer:3276.8;I-current:6553.5;pressure:255;WW-flow:0;Byte27:0;Byte28:0

Could it be the reason and what it is causing it. Is it indeed a heater-system problem or some connectivity issue between heater and mini-adapter??

[image]

[Ccollgate.zip](https://github.com/norberts1/hometop_HT3/files/2729682/Ccollgate.zip)

Thanks again for analysis and Happy New Year :)

norberts1, 2020-10-30:
It seems to be ok now, so I'll close this item.

closed 2020-10-30 by norberts1