norberts1/hometop_HT3 #11: x026A telegram

PhillyGilly, 2020-02-02:
Hallo @norberts1 Danke fúr die fantastike EMS2 infos.
I have an SM200 solar module and have been helping @Proddy to get the data from the EMS2 telegrams into MQTT.
My SM200 sent this x026A telegram in test mode.
B0 00 FF 00 02 6A 03 03 03 03 01 03 03 03 03 03 01 03
I agree with you that byte 10 represents Solar Pump but also believe that byte 4 is the cylinder selector valve VS2.  Also the values 01/03/04 are test/off/on.
This is further documented here:
https://emsesp.github.io/docs/#/SM100
I hope this helps.
mfG

norberts1, 2020-02-16:
Hallo @PhillyGilly sorry for that late answer. I was busy with other stuff.
Regarding that telegram-type: x026A I have only that Solar-Pump information at byte 10.
In my system I'm using that Bosch/Junkers MS100 modul for solar-handling connected to that main-controller CW400. From that MS100 modul I get that 0x26A like:
 B0 00 FF 02 02 6A 03 00 00 03 03 00 00 00 03 5d -->> MesID 874 with offset 2.
Keep in mind that I handle that telegrams as Message-ID's and not telegram-types (its's a bit easier to talk about) and also Junkers/Bosch internal handling is done like this.
(that calculation of message-ID's is defined in my document).
Any way, regarding that MesID 874 (FF xx 02 6A) I haven't more information and no MS200 modul.
But that payload-decoding is independent of the used modul (MS100/200 or SM100/200), only the MessageID itself is important. The difference between MS200 and MS100 is only that MS200 can handle 2 solarcircuit's and is using 2 messageID's like:
1. Solar-system: 874 (FF xx 02 6A payload) and
2. Solar-system: 875 (FF xx 02 6B payload).
Please check it out in your system, I can't.
Perhaps your SM200 is also using 2 different device-ID's for that messages like: 0x30 and 0x31 but I'm not sure.
But anyway, only the messageID is important and not the device-ID (modul) of the sender.
You can test this with the EMS-ESP softwares using not the default 'ServiceKey:0x0B' but another one like :0x0D (Modem / Esaycom) and it works if you recompile that software.
I did this check on my system already and I can run my hardware in parallel on the same heater-bus with a different device-ID :-)
Hopefully a step forward.
mfG

PhillyGilly, 2020-02-21:
Hello @norberts1

norberts1, 2020-02-22:
Hallo @PhillyGilly. Yes, that 0262 (MsgID 866) is assigned for that temperatur-values. More details you can find in my documentation in the HT3/doc-folder. See also this:
[image]
Also that Solar-Options (B,C,D,G) are defined there, but I'm not sure they are fitting to your system.
It should be 1B as you mentioned but I can't assign that senor-values: (PS1), VS1, TS1, TS2 and TS5 to that telegram. Perhaps you are able to do this.
Did you already checked your modul MS200 regarding that hardware conntection of sensors?
mfG

norberts1, 2020-10-30:
@PhillyGilly it seems to be ok now, so I'll close this item.

closed 2020-10-30 by norberts1