norberts1/hometop_HT3 #12: CR100 Setpoint messages

fwPunsher, 2020-04-13:
Hi @norberts1 ,

thechnically this is not directly connected to your software, however you seam to be the most knowledgeable guy in this field and maybe you can help me.
I am running the ems-esp module by BBQKees with the software from proddy. Using this I am able to decode the messages of my heating system (Junkers Cerastar +Boch CT100). However i havent figured out yet, how to change the mode or setpoint of the thermostat.

Do you know which message (hex codes) I would have to send to the thermostat to do this? Or could you maybe point me in the right direction to figure this out?

Best Regards

norberts1, 2020-04-16:
Hi @fwPunsher,
I'll try to help you as good as I can. 
First one question: you wrote that you have the CT100 in your system, but the issue is regarding that CR100. Do you have both controller running parallel in your system?
I don't thing so but anyway, both controllers I didn't had in my system. Currently I have that CW400 running at before that the CW100 and the FW100.
So my support is most valid for that type of controllers, but good luck that Cxyz series and it's handling is very common.
So, if you are able to send commands with your ems-esp module to your heater-system then let's test it with some messages. As always it is like: try and check the result.
The messages used for that Cxyz controllers differs from that used at Fxyz-serie. But that basic-handling is the same!
The first message for test is to switching from 'auto' to 'manual' mode. This will be displayed on your controller. Please check the results for all test sequences.
1. Auto to manual - mode (CRC is calculated by software)
`8B 18 FF 00 01 B9 00 <CRC>`  
2. Back from manual to auto-mode
`8B 18 FF 00 01 B9 FF <CRC>`
3. Setup Setpoint temperature for 'Comfort2' to 21.5 degree
`8B 18 FF 02 01 B9 2B <CRC>`  <<-- 2B is 21.5*2:= 43dez or 2B hex.
4. Setup Setpoint temperature for 'Manual' to 22.0 degree
`8B 18 FF 00 01 B9 00 <CRC>`  <<-- switching to manual - mode
`8B 18 FF 0A 01 B9 2C <CRC>`  <<-- 2C is 22.0*2:= 44dez or 2C hex.

Hopefully it works at your CT100 or CR100 or what ever it is.
I have checked it at my CW400 and it works.

fwPunsher, 2020-04-17:
Hi norbert, you were right it is a CR100 (I got confused there). 
Thanks for the messages they all work. How did you figure them out? I couldnt find the commands for the Cxy Devices nowhere on the internet. The ones for the Fxy modules are more known and already implemented in ems-esp. I tried guessing them from "HT3-Bus_Telegramme.pdf" however with no luck.
Anyways thanks, you were a great help.

proddy, 2020-04-17:
Yes, Norbert is a super-star. Still amazes me how he knows all this! I saw the mistake I made in the telegrams I sent you earlier @fwPunsher (using 2B9 instead of 1B9). I'll adjust my EMS-ESP to support CR100 this weekend. thanks both

norberts1, 2020-10-30:
@fwPunsher @proddy  it works so far and I'll close this item.

closed 2020-10-30 by norberts1