norberts1/hometop_HT3 #16: mqtt does not support lwt?

clh27, 2021-06-07:
**Describe the bug**
A clear and concise description of what the bug is.
It seems that there is no support for LWT topics in MQTT binding

**To Reproduce**
Steps to reproduce the behavior:
see mqtt sent by hometop, seems LWT is completely missing

**Expected behavior**
LWT should be set to "Online" / "Offline"

**Screenshots**
n.a.

**Additional context**
n.a

norberts1, 2021-06-12:
High, this seems to be a bug on using it in the application and not a wrong MQTT binding.
That LWT has to be set before connecting to the broker (see source-code of binding):

 `    def will_set(self, topic, payload=None, qos=0, retain=False, properties=None):
        """Set a Will to be sent by the broker in case the client disconnects unexpectedly.
        This must be called before connect() to have any effect.
        ....
`

So, I'll modify that mqtt-part in the project with a new release.

norberts1, 2021-06-24:
I'll close this issue now.

closed 2021-06-24 by norberts1