norberts1/hometop_HT3 #19: Some information is not displayed from solar heating system

Heiko-L, 2022-09-13:
**Describe the bug**
From:

sol_yield_last_hour
sol_yield_sum
sol_yield_last_day

Only sol_yield_last_hour gets populated, although the Junkers FW 120 displays the other values.

**Additional context**
The bus system consists of:
Junkers CERAPUR-ECO ZSB14-3
Junkers ISM1
Junkers FW120

norberts1, 2022-09-18:
Yes, I have seen this years ago on my FW100. Only the 'sol_yield_last_hour' - value is send with message-ID: 259.
The other values (sol_yield_sum and sol_yield_last_day) must be generated in the Fxyz-Controller itself, but never send with any messages to external moduls.
The other Controller-Types 'Cxyz' like CW100 or CW400 send all three informations with other message-ID's to external moduls, but this message-ID's are never used by the Fxyz-Controllertypes.

Message-ID 259
[image]

Heiko-L, 2022-09-18:
Ah, I see. Thank you🤗

closed 2022-09-19 by Heiko-L