norberts1/hometop_HT3 #20: T_soll gets populated twice

Heiko-L, 2022-09-20:
```
Local_date_time     | UTC        | T_soll | T_ist | T_speicher | C_betriebs_zeit  | C_brenner_ww | V_WW_einmalladung | V_WW_desinfekt | V_WW_erzeugung | V_WW_nachladung | V_WW_temp_OK | V_lade_pumpe | V_zirkula_pumpe | V_spare_1 | V_spare_2 | hexdump
2022.09.20 13:20:27 | 1663672827 | 48     | 48.5  | 48.5       | 783.416666666667 | 410          | 0                 | 0              | 0              | 0               | 1            | 0            | 0               | 0         | 0         | 27_0 :WW :90 00 1b 00 30 00 b0 00 23 00 10 00 90 00 2b 00 30 00 b0 00 33 00 3b 00 10 00 90 00
2022.09.20 13:20:30 | 1663672830 | 15     | 48.5  | 48.5       | 783.416666666667 | 410          | 0                 | 0              | 0              | 0               | 1            | 0            | 0               | 0         | 0         | 52_0 :WW :88 00 34 00 0f 01 e5 01 e5 a1 00 00 03 00 00 b7 9d 00 01 9a 00 3f 00
```

T_soll is populated with "Soll-Temperatur Warmwasser (Zeitprogramm gesteuert)" and with "Sollwert Warmwasser-Temperatur".
When rendering T_soll in a diagram it looks really ugly.
I don't know where the 48 is coming from either (I can only imagine that this is the value set at the heater, but that should be 51_2?).

TIA,
Heiko

norberts1, 2022-09-20:
@Heiko-L I don't know where the 48 is coming from either (I can only imagine that this is the value set at the heater, but that should be 51_2?).
Yes, it seems that MsgID: 27_0 sending this information: 51_2 too.
But the value-assignment is to the same placeholder and the meaning differs.
At my current heater-system only MsgID: 51_0 and 52_0 are send, the T-Soll Max value is send with MsgID: 51_2.
see picture:
[image]
This DHW message-handing is very dependent from the heater-system like:
1. creating WW boiler-internal or not
2. using external sensors included in the IPM-modul
3. etc.
Please tell me which design your heater-system has.
If you create and post one binary-logfile it would be good for checking and test.

Heiko-L, 2022-09-21:
Take a look [here](https://github.com/norberts1/hometop_HT3/issues/18#issuecomment-1253301971)

I also wonder where "T-Ist  (Raum)" from HK1 comes from in my setup...

norberts1, 2022-09-21:
To handle this error some software-modifications are required, see #18 

@Heiko-L 
I also wonder where "T-Ist (Raum)" from HK1 comes from in my setup...
Edit: Could it be that this is a temperature sensor from the FW120 which would serve as room temperature sensor if the FW120 is wall mounted? So now it's displaying internal heater temperature because the FW120 rests in the heater.

Yes, this is the reason.

Heiko-L, 2022-09-23:
After editing the blacklist the "Sollwert Warmwasser-Temperatur" no longer populates "T_soll" until now.

norberts1, 2022-09-23:
>  ...no longer populates "T_soll" until now.

At least MessageID 52_0 with that T-Soll value should be send and displayed in that HT3_Analyser.py as shown above.
**Or is this value not available?**

Please check also that msgID 51_0 -->> T-Soll max in that HT3_Analyser.py. 
You can force this message, if you set that WW-temperature potentiometer build in your ZSB -boiler to another value.
see picture:
[image]

Heiko-L, 2022-09-26:
52_0 is received and populates T-Soll.
27_0 is received.
For 51_0 I'll have to investigate....

norberts1, 2023-03-14:
#20 fixed.

closed 2023-03-14 by norberts1