norberts1/hometop_HT3 #21: Alerts to external instances

norberts1, 2022-09-27:
**Describe the enhancement**
System-alerts should be available for external instances.

**Expected behavior**
Heatersystem-alerts should be send out to available interfaces or should call (callback) handlers doing this.
At least the current error-code should be send out.

**Additional context**
A mapping between error-code and describing text would be nice.

NilsRo, 2022-09-27:
Still had ErrorCode 40 which was shown as this in EasyRemote App. But I cannot find the code for too less pressure. I found some entries with E10 as code. It is is Hex...but it is not listed in the manual.
[FW200 Error Codes.pdf](https://github.com/norberts1/hometop_HT3/files/9659420/FW200.Error.Codes.pdf)

Also here is a good overview with pressure E9 but I am not sure as I am refilling everything regularly now.

norberts1, 2022-10-01:
Bitte ein Logfile der Heizungsanlage erzeugen, die diese Fehlermeldung enthält und hier posten.
Damit kann dann die Software-Änderung geprüft werden.
Logfile erzeugen mit:

1.  ins Verzeichnis wechseln: ~/HT3/sw auf dem RPi.
2.  Start des Logging mit: ./ht_binlogclient.py irgendein_logfilename.log

Das Logfile ist dann zu finden unter: ~/HT3/sw/var/log/irgendein_logfilename.log

Das Logfile sollte die Fehlermeldungen enthalten und nicht grösser als 2 Stunden generiert worden sein.
PS: der Prozess: ht_proxy.py muss dafür konfiguriert sein und laufen (Default-Einstellung des Projektes).

NilsRo, 2022-10-08:
@norberts1 : Wird gemacht. Bin erst am 14. wieder daheim und dann lass ich mal etwas Druck ab. :-)

NilsRo, 2023-03-19:
Hi Norbert. After upgrading I checked the MQTT messages (now using node red instead of a not working homeserver plugin) and saw a ch_errorcode  and ch_causecode topic. Do this topics publish the errors shown by FW200 display? So then there is no need to change something as the error coded could still be used?

norberts1, 2023-03-19:
> Do this topics publish the errors shown by FW200 display?

Yes

> So then there is no need to change something as the error coded could still be used?

I'm not sure, cause I didn't find any error-codes for 'pressure to low' etc.
The heater itself has to check this pressure and if it is to low must send an error to the FW200.
I can't really check, cause my heater hasn't build in a pressure-sensor.

Perhaps you can check this on your system and watching the FW200-display for any error-presentation.

NilsRo, 2023-03-20:
Will report back on 27. as my heating engineer has to add some heat meter and the pressure is still not that high actually.

NilsRo, 2023-03-30:
So, my workman should leave the heater on during draining the pipes but anyway...he doesn't. So I played around a little with other errors like switching of the solar module which created an alert in my EasyRemote app and on the FW200 screen. I expect the same in my MQTT topics bit ch_errorcode and ch_causecode stays on 0.

I created a binary protocol attached. It started with the error, I put the solar modul back online and error is gone and then I switched the solar module off again.

Should I see 12-774 also in MQTT error fields?
[ht_binlog.log](https://github.com/norberts1/hometop_HT3/files/11116128/ht_binlog.log)
[image]
[image]
[image]

norberts1, 2023-04-23:
Thank's for the logfile and work.

> Should I see 12-171 also in MQTT error fields?

Not yet with the current release.
Cause they are using a different telegram to send error-code and cause-code for systems with Fxyz-controller and this has to be implemented.
With correct implementation you should get mqtt-messages for the above case:
ch_errorcode = 12
ch_causecode = 171

The error-code handling differ between controller-type (Fxyz/Cxzy), byte-sizes and telegrams.
It's a bit of nightmare :-)

If I have a new testable script I'll place it/them on this issue for your testpurposes.

norberts1, 2023-04-27:
Attached you'll find one tar-file for updated modules regarding this and other issues.

1. errorcode and causecode should be send by mqtt also for Fxyz controllers.
2. pressure-value is now send and GUI-displayed if the system has a pressure-sensor included.
3. solar-optimazation factors are displayed and send if solar is available.

After untar and replacing the old with new modules is strongly required to recreate (delete the old one's) databases for sqlite and rrdtools.

Please check it out and let me know of the result.

[issue21_test_1.tar.gz](https://github.com/norberts1/hometop_HT3/files/11341924/issue21_test_1.tar.gz)

NilsRo, 2023-04-28:
Installed the updates remotely and pressure looks great. Solar-Optimization is also shown with values different from 0 but I am not familiar with that. For testing the error codes I have to be back home which should be in 3 weeks. Will test it soon then...

NilsRo, 2023-05-22:
Tested now with an error and it shows up in MQTT. But only one error code, In the iOS App I can see two errors in parallel.

 
[image]
[image]
[image]
[image]

norberts1, 2023-06-24:
That errocode: 12594 is send as a number with mqtt, but should be send as a string:12
(12594 )dec := (3132)hex as ascii -->>  "12"
I have to modify this special mqtt-output to send this value as a string.

The other values: 'D5-15' are additional infos and not send as ch_errorcode and ch_causecode and I'll not support them to be displayed.

NilsRo, 2023-06-25:
https://www.heizungfehlermeldung.de/fehler.php?merk=6&ketel=133&storing=3179&subcode=0

Hier stehen übrigens die Fehlercodes. Da steht D5 auch drinnen. Ist vielleicht ganz spannend.

norberts1, 2023-07-02:
I have made a new testrelease for this issue#21 and issue#27.
Please check it out on your system. Your current databases (if already available) must not be updated.
This software includes:
#21: ch_errorcode now send as string with mqtt also to homeassistant.
#27: temporary logfiles located now in /tmp.
It works so far on my system, but only one thing I have to investigate -->> on my homeassistant dashboard the errorcode will NOT be displayed as a string. If the errorcode is a number then it is OK and displayed.
Attached you can find my mqtt-logfiles and the errorcode-strings are send with mqtt also to the homeassistant. Perhaps you have any idea and can fix this detail in the homeassistant (if you use this).

Logfile:
[mqtt_errorcode_testlogs.tar.gz](https://github.com/norberts1/hometop_HT3/files/11929585/mqtt_errorcode_testlogs.tar.gz)

Testsoftware:
[issues21_27_test_1.tar.gz](https://github.com/norberts1/hometop_HT3/files/11929587/issues21_27_test_1.tar.gz)

Please check this and if this fit to your purposes I'll make a new sw-release and close that items.

NilsRo, 2023-08-08:
Works flawlessly. But I cannot help with homeassistent. Using node.red and KNX. But it looks like an automatic conversion....I using some ESP32 devices and there numbers and strings are automatically converted to numbers if possible. Perhaps this is the case here also...

closed 2023-10-12 by norberts1