norberts1/hometop_HT3 #22: Pressure of water is not available

NilsRo, 2022-09-28:
It would be nice if some pressure measures could be added to MQTT interface at least the water supply line. I have to refill it recurrently and you can use it to avoid errors. This value is also visible in EasyControl App I also use...

norberts1, 2022-09-28:
> It would be nice if some pressure measures could be added to MQTT interface ...
Not all boilers / heater-devices have this pressure-value available. In that cases nothing will be send if invalid values are detected.

NilsRo, 2022-09-28:
Question or comment how you plan to implement it?

norberts1, 2022-09-28:
MsgId 24_17_0 holds that information for pressure and if valid it can be send out with MQTT and stored to the dbases.
See picture from telegram-documentation:
[image]

NilsRo, 2022-09-28:
That I also found but it seems actually not available in MQTT but the MB-LAN2 must get it from somewhere. I am new to MQTT in general so perhaps I missed something. What is the topic I should look for?

norberts1, 2022-09-28:
>  but the MB-LAN2 must get it from somewhere.

It get's this with above mentioned message, what else. It's connected to the heater-bus.

This pressure-value will be avaiable in any of that implemented and enabled interfaces: sqlite, rrdtool and mqtt.
The new placeholder must be defined, but this will be done in that xml-configurationfile.
I'll think you are using that sqlite-interface, right?

NilsRo, 2022-09-28:
Ich schreib mal auf Deutsch, verstehe den aktuellen Stand nicht. 

Ich nutze nur rrd und MQTT. Dort sollte das der Druck schon unter einem eigenen Topic verschickt werden wenn die Anlage diesen sendet? Wenn nein, muss ich was tun oder du im genannten XML? Wenn ja, wie heißt das Topic? Ich habe kein passendes gefunden...

norberts1, 2022-09-29:
Die Auswerte-Software muss noch erweitert werden. Deshalb ist dieses Issue auch als 'enhancement' eingestuft und nicht als 'bug'.
Wird im nächsten Release enthalten sein.

NilsRo, 2023-03-18:
Hi, any updates to the pressure integration? The only thing that is missing for me and the perfect smart home integration.

norberts1, 2023-03-19:
Yes, in the current release there is decoding of pressure included. You can find this in the moduls: ht_discode.py and gui_worker.py
But for two reason this is not yet activated:
1. The databases have to be extended for new values (sqlite and rrdtool) and not everyone want to do this yet.
2. I haven't got any logfile for testing this value in my environment (see issue: #21).
Please create one logfile, then I can test this and will create a new release.

NilsRo, 2023-03-19:
Attached the log. Hopefully it is included. 
I thought you need the log for the alert only. 
 
[image]
[pressure.log](https://github.com/norberts1/hometop_HT3/files/11011674/pressure.log)

NilsRo, 2023-05-22:
Working fine, thanks!

closed 2023-05-22 by NilsRo