norberts1/hometop_HT3 #23: ht_proxy.py never recovers from USB disconnect

joed74, 2022-11-02:
I am using a pitiny adapter with an ttl USB converter.
I am using an udev-rule so that the USB device always shows up as /dev/heatronic3 symlink.

If i disconnect the USB port and reconnect it, ht_proxy just freezes and i must restart ht_proxy and ht_collgate.

Would be nice, if ht_proxy could try to reconnect every second or so...

norberts1, 2022-11-03:
I'll check this. Best to throw an exception and try to reconnect that interface n-times.

norberts1, 2022-11-23:
@joed74 
Please update your system with that modified files:
1. ~/HT3/sw/lib/ht_proxy_if.py 
2. ~/HT3/sw/lib/ht_release.py

After this update stop and restart that processes with:
1. sudo systemctl stop ht_collgate.service
2. sudo systemctl stop ht_proxy.service
4. sudo systemctl start ht_proxy.service
5. sudo systemctl start ht_collgate.service

Above mentioned problem is fixed with this modification.
Please check it out at your system.
Keep in mind that:
After disconnection and reconnecting the USB port the ht_proxy.py will restart and reopen that serial USBport connection and RESTART also that TCPServer.
So you have to reconnect all previously connected proxy-clients.

joed74, 2022-12-15:
> Above mentioned problem is fixed with this modification.

ht_proxy now connects wonderfully again, but the basic problem that no more graphs are drawn and no more data is transmitted via mqtt is still there as before ;(

> So you have to reconnect all previously connected proxy-clients.

What is the difference between:

before: USB disconnect, reconnect ht_proxy_if.py stops working, ht_collgate.py stops working -> no graphs, no mqtt

now: USB disconnect, reconnect ht_proxy_if.py still working, ht_collgate.py stops working -> no graphs, no mqtt

Isn't it possible to hold the client connections up and running an only reconnect to the USB data source?

norberts1, 2022-12-27:
> Additional observation: The status in the MQTT remains online, although no data has been sent for half an hour...

The MQTT-status 'online' is assigned to the MQTT-client availabilty itself and not to any data-transfer. The 'ht_collgate.py' and it's included mqtt-client are still running, so that MQTT-status must be 'online'.

> Isn't it possible to hold the client connections up and running an only reconnect to the USB data source?

Yes, this should be possible and a solution. But keep in mind also that socket- and thread-handling is my aim to be improved.
This takes a bit time, so in the meantime don't disconnect that USB-port and use it as a stable connected serial interface.