norberts1/hometop_HT3 #24: Unable to set desired temperature using mqtt (home assistant)

toams, 2022-12-30:
Hello,

I'm trying to use home assistant to change the desired room temperature and also turn heating on/off (heizen/sparen?)
However i cannot seem to figure out the topic and payload i have to use.
In the manual i can find following command:
`mosquitto_pub -d -h <host IP-Adr.>   -t set/hometop/ht/hc1_Tdesired -m "21.5,heizen"`
But running it on the command line doesn't seem to work (I  changed ` <host IP-Adr.>` to my mqtt brokers ip). 
Running: `mosquitto_sub -v -h rpi3.local -p 1883 -t 'rpi3/hometop/ht/hc1_Tdesired'` gives correct results so the mqtt broker is working.

I noticed that `rpi3` is added to the topic when using homeassistant. I tried adding `rpi3` to the topic on various places, but nothing seems to work.

So maybe there is a bug in the ht_2hassio.py-daemon? Or maybe I am doing something wrong. Also maybe the answer be found in the docs, but my German is not very good, so i might have missed something.

norberts1, 2023-01-01:
> I noticed that `rpi3` is added to the topic when using homeassistant.

This `rpi3` addon is the name of your RPi and added in front of the root-topic per default.
The resulting topic is then like this one: `rpi3/hometop/ht/hc1_Tdesired`.
In some cases it is useful to have this information to separate topics send by different machines.
But in most cases it is unused and not required and should be disabled per default.

To disable this addon modify that mqtt-configuration file used by `ht_collgate.py` and `ht_2hassio`.
You have to restart that processes again.

 config-file: ~/HT3/sw/etc/config/**mqtt_client_cfg.xml**
 modification:
    <mqtt_client>
       ....
        **<device_id></device_id>**
       ....
    </mqtt_client>

Please check this modification on your system with that following command (for **Fxyz** controllers):
  `mosquitto_pub -d -h <host IP-Adr.>   -t set/hometop/ht/hc1_Tdesired -m "21.5,heizen"` 
or for **Cxyz** controllers:
  `mosquitto_pub -d -h <host IP-Adr.>   -t set/hometop/ht/hc1_Tdesired -m "21.5,comfort2"` 

Also you can enable the debug-mode for mqtt-messages with:
 < logging>
 .....
 < loglevel>**DEBUG**</ loglevel>
 </ logging>

I'll disable this topic-addon per default and check it again in my heater-system together with homeassistant.

toams, 2023-01-02:
Thanks for your quick response. Removing localhost from <device_id> made it work. If I have some more questions in the future, what is the best way to ask them? I noticed the German Mikrocontroller forums, but I don't speak German. 

Thanks again for your time and keep up the good work!

norberts1, 2023-01-04:
> If I have some more questions in the future, what is the best way to ask them?

I'll think this is the right place here located by the project using English language.
If you have a new topic (also for questions) then you can open a new issue.

closed 2023-10-12 by norberts1