norberts1/hometop_HT3 #27: 2 Logfiles are not located in log folder

NilsRo, 2023-05-25:
**Describe the bug**
I setup an read-only filesystem and used HT3 to send MQTT messages. For this I had to put var/log and var/run to a tmpfs. During investigation why it is not running fine I found 2 logfiles still not created in the logs folder. cdata.log and cdb_rrdtool.log. solved this via symlinks but it should be moved to log folder also. (a rsyslog integration would be nice also ;-) )

**To Reproduce**
Steps to reproduce the behavior:
1. Go to folder HT3/sw
2. See logfiles

**Expected behavior**
All logfiles should be located in the log folder or provided in the central log via rsyslog  https://gist.github.com/danielkraic/a1657f19bad9c158cbf9532e1ed1503b

**Additional context**
Avoids broken SD Cards, etc.

norberts1, 2023-06-06:
cdata.log and cdb_rrdtool.log are created as defaults on startup (if yet no logfile-names are available).
Normally they have no content if the configuration-file reading is successful.

So my fix will be to place this files into the temp-folder.

NilsRo, 2023-08-08:
Works flawlessly

closed 2023-08-08 by NilsRo