norberts1/hometop_HT3 #28: SW-enhancement for different Solaroptions

norberts1, 2023-07-16:
**Describe the bug**
The decoding for different Solar System-Konfigurations (SolarOptions) should be implemented.

**To Reproduce**
Systems with EMS2 Solarmodul MS200 can have many different configurations and telegrams.
See logfiles on URL: https://www.mikrocontroller.net/topic/324673?page=single#7454290

**Expected behavior**
Decoding for different solar options is done and send to external interfaces.

Negruej, 2023-07-16:
Hi Norbert,
 - for option "B":  bin log file should contains info for PS1 and VS2 (see also time stamps in attached xls file )
Regards,
Jürgen

[log-16072023.xlsx](https://github.com/norberts1/hometop_HT3/files/12065051/log-16072023.xlsx

[cw400-log-16072023.tar.gz](https://github.com/norberts1/hometop_HT3/files/12065050/cw400-log-16072023.tar.gz)

Negruej, 2023-07-17:
Hi Norbert,

telegram 868_9 - byte6 should be the solar pump PS1 power ( in case of PWM used)

Regards,
Jürgen

Negruej, 2023-07-18:
Hi Norbert,

info for solar option "J" ( reload SP1 to SP3 )

Regards,
Jürgen

[cw400-log-18072023.tar.gz](https://github.com/norberts1/hometop_HT3/files/12082111/cw400-log-18072023.tar.gz)
[cw400-log-solar-option-J.xlsx](https://github.com/norberts1/hometop_HT3/files/12082112/cw400-log-solar-option-J.xlsx)

norberts1, 2023-07-18:
> telegram 868_9 - byte6 should be the solar pump PS1 power ( in case of PWM used)

Yes it is.

Thanks for that logs. 
I have to make many extensions for that solar options and decoding.
- At first I'll add some debug-outputs for you. Then you can check it in parallel with your system.
- Next step will be to show the results on the Analyser Display in realtime.
- At at least all interfaces (mqtt, rrdtool, sqlite) have to be updated.

But this is what I want to do, and good luck: you can create the log-files and will test the results :-)

Attached you can find a commented logfile with the assignment of temperature sensors (TSxy) to the bytes in the telegramm-streams (cw400-log-16072023a.log_comment_part1.zip). 
Please check it for the right assignment.
[cw400-log-16072023a.log_comment_part1.zip](https://github.com/norberts1/hometop_HT3/files/12083310/cw400-log-16072023a.log_comment_part1.zip)

Also I have made an update for the telegram documentation with a first assignment of temperature sensor names to the bytes in telegramm ID:866.
Must not yet be OK, but it seems to fit to your system. Please check it.
[image]

Negruej, 2023-07-18:
- in file "cw400-log-16072023a.log_comment_part1.zip" values for TS1, TS2, TS3, TS4, TS5, TS10 and TS11 correct compared to my system with option 1ABJ
 - for the docu I can confirm 866_0_0 and 866_2_0. 
 - would suggest: 866_8_0 = "Solar Option A: Speicher1 mitte, TS3", 866_10_0="Solar Option A: Return-Temperatur, TS4",    866_16_0 = "Solar Option B: Speicher2 unten, TS5", 866_24_0 = "Solar Option J: Speicher3 unten, TS11"

norberts1, 2023-07-20:
The assignment of Solar Option letters to the bytes in MesID:866 and others are done by Junkers itself. 
I found and used this from there older documentation.
But never mind, some of the TSxy names are used twice and more for different Solar-Options.
Only TS1, TS2 and TS7 have fixed assignment to Solar-Options letters. The others are more or less used in different configurations.
Common interface overview for MS200:
[image]

I have made a new software testversion for you to check it out on your system.
It's important to **recreate the used databases**, cause there are new logitems used for the solar-part.
Please enable the DEBUG flag in the configuration-file to get detailed info on messages.
In the HT3_Analyser the additional solar-measurements are visible only in the **'solar'-view** and not yet in the system-view.
The tar-file holds the complete project-files in the latest version for tests.

Please check it out on your system.
[ht_sw_rev_0_6_test1.tar.gz](https://github.com/norberts1/hometop_HT3/files/12105718/ht_sw_rev_0_6_test1.tar.gz)

Negruej, 2023-07-20:
I use a distributed deployment ( proxy on RaspiZW / client on Proxmox CT ) do I have to update both or only the client ?

norberts1, 2023-07-20:
> do I have to update both or only the client ?

Only the **client**, cause no change in the ht_proxy-sw and message-transfer.

Negruej, 2023-07-20:
it works immediately after update, alle the missing values now shown in the analyzer window.
Thanks a lot, great work.

Negruej, 2023-07-21:
could you please verify that my findings are correct: Telegramm 868_14 Byte6 (or 868_0 Byte 20) contains the value for PS7 ? Currently it shows always "Aus" 
[image]

norberts1, 2023-07-21:
> Telegramm 868_14 Byte6 (or 868_0 Byte 20) contains the value for PS7 ? 

It contains the: 'Required Reload Pump Power' for the PS7.
I have used the flag 874_1_2 on 874 for PS7, but this one is **not set**.

But it could be the 874_11 used for PS7.  If 874_11 = **4 -> PS7 running** and 874_11 = **6 -> PS7 off**.

Attached you'l find an update of ht_discode.py with above mentioned modification. 
Extract the tar-file and move the content to ~/HT3/sw/lib.
[ht_discode_20230721.tar.gz](https://github.com/norberts1/hometop_HT3/files/12131477/ht_discode_20230721.tar.gz)

Negruej, 2023-07-21:
with the modified "ht_discode.py" show the analyzer window now the correct state from PS7.
The attached bin- and debug-log file contains 22:31=ON / 22:32=OFF and 22:40=ON / 22:41=OFF for your reference.
[cw400-ps7-bin.log.tar.gz](https://github.com/norberts1/hometop_HT3/files/12134746/cw400-ps7-bin.log.tar.gz)
[debug-log-PS7.tar.gz](https://github.com/norberts1/hometop_HT3/files/12134747/debug-log-PS7.tar.gz)

norberts1, 2023-07-26:
@Negruej 
I have made a new release for tests on your system. Modifications are:
- Systemview and Solarview updated.
- many new solar logitems and name-changes done.
- size of rrdtool-database reduced to hold at least 2 years of data.

For above mentioned changes it is **required to recreate the databases** with call: .create_databases.py

Please check it on your system and hopefully it runs as good as the previous test-release.
All project sw-parts are included in the tar-file.
[ht_sw_rev_0_6_test2.tar.gz](https://github.com/norberts1/hometop_HT3/files/12176436/ht_sw_rev_0_6_test2.tar.gz)

Negruej, 2023-08-02:
Hi Norbert,

*_test2.* works just like the previous version. I can not find problems in this version

norberts1, 2023-10-01:
@Negruej 
I have made some modifications on rrdtool grafic-draw for matching to your system.
There is an additional draw for your special system filled with temperature- and pump-data.
Attached you can find the first draft on this output:

[image]

If this matches to your requirements I'll create a new release.

norberts1, 2023-10-01:
@Negruej 
During tests with that HT3_Analyser using your logfiles I found one Displaycode (error): A11 and CauseCode: 3061.
Perhaps there is something wrong configured or faulty on your system.
See attached grafic-output:
[image]

norberts1, 2023-10-12:
@Negruej  New release made, please check it.

Negruej, 2023-10-13:
@norberts1 I have installed the new release on a new separate system with the provided script. 
It works instantly after set the ip from proxy and MQTT broker. Page from second "Speicher" now available.
Here the example from today:
[image]
We have changed the setup to "1BDHI" mainly to activate option "H".
[image]

VS1 is now changed to VS3. 
Therefore I did not expect that all values on the display now correct. I will generate new logs and update this here later this weekend. 
Thks for your effort and help ..

Negruej, 2023-10-16:
@norberts1 Could you please check the value of TS8 on the second solar graph. I'm not sure that the value of TS11 was shown there in my first setup.
I have generated bin log from 3:07 - 3:19 pm  today and also take pictures from CW400 to show the value's.
To control the function of the "solar heating support" would it be possible to show also VS3 ? ( in % )
Until now it works perfect and is a great help to tune my setup.
Thanks and regards,
Jürgen
[bilder.zip](https://github.com/norberts1/hometop_HT3/files/12918440/bilder.zip)
[binlog03-07bis03-19.log](https://github.com/norberts1/hometop_HT3/files/12918433/binlog03-07bis03-19.log)

Negruej, 2023-10-21:
@norberts1 I have verified TS4 and TS8 on MS200:
- If I disconnect TS4 Telegramm 866_10 show value 0x8000 = sensor error: open
- If I disconnect TS8 Telegramm 866_22 show value 0x8000 = sensor error: open, on reconnect id changed to value 0x00c0 ( 19,2°C) and the CW400 show 19°C for TS8
[image]

Negruej, 2023-10-21:
@norberts1 vor Option "H" (VS3)
 - mixerposition is in telegram 868_15, Diagnose on CW400 show exact same value in % under "Mischerposition"
 - on time the mixer is moving in direction "open" telegram 868_1 show value "64", Diagnose on CW400 show "Mischer öffnen" Ja
 - on time the mixer is moving in direction "close" telegram 868_0 show value "64", Diagnose on CW400 show "Mischer schließen" Ja
If you need more bin or debug logs pls. request.
Regards,
Jürgen

norberts1, 2023-10-25:
@Negruej Thank's for your logfile and investigation.
I have implemented this mixer-position 868_15 in my testenvironment, see picture.
The assignment is currently to the 'Solar'-part of the 'Analyser', but I'm very unsure that's the right place.
Cause, the 'Heizungsunterstützung' is foreseen for one of your two heating-circuits (I'll thing it's circuit 2).
And the logical assignment should be then: heating-circuit 2 for the mixer-position,  cause there is already one placeholder for it.
See pictue:
[image]

It would be very helpful to get one extended logfile with data including also heating-circuit2 and more information on your currently configured and used system.

Negruej, 2023-10-26:
@norberts1 here my findings:
the "Mischer" in HK1 or HK2 is installed depends you have "underfloor heating" or "radiator" installed there. Here is one example of mixed setup, this will be my final setup (vor now I have only HK1 active): 
[hk1+hk2.pdf](https://github.com/norberts1/hometop_HT3/files/13180005/hk1%2Bhk2.pdf)

The "Heizungsunterstützung" can be implemented with VS1 (as option "A" on SP1 or "D" on SP2) as a switch. 
If temp on TS3 > TS4 then VS1 = 1, the RL use SPx to 100% until TS3 < TS4 then VS1 = 0 means SPx is bypassed in RL
If a Mischer is installed there, option "H" is used in addition to "A" or "D". The system will now try to bring the return temp on TS8 to the required VL temp. with Mischer position between 0 and 100% until  TS3 < TS4 ( no heat avail in SPx )
Here the prinziple drawing for the "Heizungsunterstützung":
[VS3.pdf](https://github.com/norberts1/hometop_HT3/files/13180412/VS3.pdf)

With this, the VS3 could be stay in solar ... (in my opinion and that doesn't have to be correct as all above)
 Thanks for your effort and help +Regards,
 Jürgen

norberts1, 2023-10-27:
@Negruej 
> With this, the VS3 could be stay in solar ...

Yes, cause the hybrid buffer is used for the hole system independent how many heating-circuits are connected.
I have made a sw-update to check it out at your system.
Database must not be new created, cause I used a spare-name for the new 'mixerposition VS3'-logitem.
The rrdtool-drawing has now this VS3-value included. Please check it out.

[ht_sw_update_0_7_x.tar.gz](https://github.com/norberts1/hometop_HT3/files/13192020/ht_sw_update_0_7_x.tar.gz)

Negruej, 2023-10-29:
@norberts1 copy the files from above and restart display TS8 and VS3 now.
[image]
Thanks + Regards,
Jürgen

closed 2023-11-11 by norberts1