norberts1/hometop_HT3 #32: error code  50-121 is not handled

NilsRo, 2025-02-18:
**Describe the bug**
I got a heater error send by Junkers App and MB-LAN2 but HT3 does not recognize it.

**To Reproduce**

1. Create error 50-121   ;-)
2. Will not be send by HT3 to MQTT Interface

**Expected behavior**
Error Code should be shown in ch_errorcode and ch_causecode

**Screenshots**
[image]
[image]
[image]

**Additional context**
Newest release from today installed

Binaer-Log attached
[binlog_for_error.zip](https://github.com/user-attachments/files/18845572/binlog_for_error.zip)

NilsRo, 2025-02-19:
**Update**: today it reached MQTT. Would it possible that the error is send only once??? But sounds strange...at this time my heating engineer checked it. The error was shown continuously in the Junkers App. 

1737902044614725722 0
1739882333793382201 0
1739962142550632486 121
1739962408043249815 0
1739963149442812414 121
1739963625333174004 0

norberts1, 2025-02-21:
> today it reached MQTT. Would it possible that the error is send only once???

Yes, all mqtt-messages are send only once. Any item must change the value to send it once again.
This behavior is changeable with configfile: mqtt_client_cfg.xml

You have to change that value:
        <Publish_OnlyNewValues>**True**</Publish_OnlyNewValues>  
from True to **False**
After changing the config you have to restart the app as always.
Then all items are send with mqtt independent from there values.

The reason for this configuration-value: **True** is the amount of mqtt-data send to the broker.
You have to check your system for overload or something strange behavior after changing the config.

> Binaer-Log attached

I have checked your binary-file and there is no display- or cause-code in that messages: 24 and 190.

> 1737902044614725722 0
> 1739882333793382201 0
> 1739962142550632486 121
> 1739962408043249815 0
> 1739963149442812414 121
>1739963625333174004 0

What type of messages this are ?

NilsRo, 2025-02-21:
That are my MQTT logs out of my influx db. The day error 50-121 occurs first there is no entry in the MQTT log. I updated HT3 to newest version but errorcode and causecode was still shown as 0 in contrary to Junkers App. The first MQTT message was shown during the repair.

If there is no entry for error 50 in the binaer log this would explain why HT3 does not recognize it. But it is strange that the MB-LAN2 seems to have it stored. Anyway, I will have an eye on it if the next error occurs.